unitreerobotics/unitree_lerobot #2: `AttributeError: 'NoneType' object has no attribute 'num_envs'` when running `eval.py`

aruiplex, 2024-12-30:
Description
I encountered an issue when attempting to run the `eval.py` script for evaluation. The error occurs when using a pretrained policy stored locally. The script fails with an `AttributeError`, and it appears that the environment object (`env`) is `None`.

---

### Steps to Reproduce
0. Train the model by using the steps in the README.

1. Prepare a pretrained policy saved at the following path:
   ```
   /data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/outputs/train/2024-12-26/15-23-22_real_world_diffusion_default/checkpoints/500000/pretrained_model/
   ```
2. Run the following command:
   ```bash
   python lerobot/scripts/eval.py --pretrained-policy-name-or-path "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/outputs/train/2024-12-26/15-23-22_real_world_diffusion_default/checkpoints/500000/pretrained_model/"
   ```

---

### Expected Behavior
The script should successfully initialize the environment, load the pretrained policy, and proceed with the evaluation.

---

### Observed Behavior
Instead, the script produces the following error:

```plaintext
WARNING 2024-12-30 22:41:56 pts/eval.py:526 The provided pretrained_policy_name_or_path is not a valid Hugging Face Hub repo ID. Treating it as a local directory.
INFO 2024-12-30 22:41:56 on/logger.py:39 Output dir: outputs/eval/2024-12-30/22-41-56_real_world_diffusion
INFO 2024-12-30 22:41:56 pts/eval.py:480 Making environment.
INFO 2024-12-30 22:41:56 pts/eval.py:483 Making policy.
Loading weights from local directory
Traceback (most recent call last):
  File "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/lerobot/scripts/eval.py", line 581, in <module>
    main(
  File "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/lerobot/scripts/eval.py", line 494, in main
    info = eval_policy(
  File "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/lerobot/scripts/eval.py", line 241, in eval_policy
    n_batches = n_episodes // env.num_envs + int((n_episodes % env.num_envs) != 0)
AttributeError: 'NoneType' object has no attribute 'num_envs'
```

---

### Additional Context
- **Script Path:** `lerobot/scripts/eval.py`
- **Pretrained Policy Location:** A local directory containing the checkpoint.
- **Environment Initialization:** It seems the `env` variable is not properly initialized. Looking at the code, the issue might be in the function responsible for creating the environment (`make_environment` or equivalent).

---

### Environment Details
- **OS:** Ubuntu24.04
- **Python Version:** Python 3.10.15

### Additional Information

If I run this:

```bash
python lerobot/scripts/eval.py --pretrained-policy-name-or-path "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/outputs/train/2024-12-26/15-23-22_real_world_diffusion_default/checkpoints/500000/pretrained_model/" eval.n_episodes=50   policy=diffusion_unitree_real_g1    env=unitree_real_g1     dataset_repo_id=UnitreeG1_DualArmGrasping
```

There will be another error:

```text
WARNING 2024-12-30 22:47:44 pts/eval.py:526 The provided pretrained_policy_name_or_path is not a valid Hugging Face Hub repo ID. Treating it as a local directory.
Traceback (most recent call last):
  File "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/lerobot/scripts/eval.py", line 581, in <module>
    main(
  File "/data/Hypothesis/proposition/unitree_IL_lerobot/lerobot/lerobot/scripts/eval.py", line 469, in main
    out_dir = f"outputs/eval/{dt.now().strftime('%Y-%m-%d/%H-%M-%S')}_{hydra_cfg.env.name}_{hydra_cfg.policy.name}"
AttributeError: 'str' object has no attribute 'name'
```

Thank you for your support! Please let me know if additional information is required.

Henry-Ellis, 2025-01-03:
Sorry for the late reply. Are you using real data from the training set? Currently, we don't support verification in a simulation environment. If you have a real G1, you can refer to https://github.com/unitreerobotics/unitree_IL_lerobot/blob/main/lerobot/lerobot/scripts/eval_g1.py.

aruiplex, 2025-01-03:
Yes, I want to evaluate it in the simulation environment. I trained the model by using the dataset you provided. Do you plan to publicize the simulation script?

Henry-Ellis, 2025-01-03:
Are you training the model (act or dp)? Currently, we haven't done any simulations like the [aloha, pusht] environment in lerobot. We always test on real devices using real datasets. Also, shouldn't you be testing on real devices when using the real datasets we've made available? Is it still the VLA model you trained?

aruiplex, 2025-01-04:
Thank you for your prompt response.

I recently trained the diffusion_unitree_real_g1 policy and would like to evaluate its accuracy using a simulation environment. Unfortunately, the G1 robot has not yet been delivered to my organization. Therefore, I plan to rely on simulations to test and familiarize myself with the policy.

From the promotional videos I’ve watched, it appears that the G1 robot is featured in simulated environments. I would appreciate it if you could guide me on how to set up and use such a simulation to effectively test the policy and gain a better understanding of its functionality.

Thank you for your assistance.

Henry-Ellis, 2025-01-04:
Currently, we do not have a simulation environment for imitation learning tasks because the strategies trained in the simulation have little reference value for strategy execution in the real environment. We focus more on the performance in actual tests (only for imitation learning strategies). Therefore, I did not include the g1 operation task in the simulation. If we have time later, we can consider adding a simple task simulation test case. The g1 you saw in the simulation should be the motion control training under value reinforcement learning. If you want to try, there is a very good project you can refer to: https://github.com/haosulab/ManiSkill. This is an extremely excellent project. If your focus is on model strategies, you can try https://github.com/haosulab/ManiSkill/tree/main/examples

closed 2025-01-09 by Henry-Ellis