unitreerobotics/unitree_lerobot #4: How should the dataset be recorded?

ltf1001, 2025-02-11:
This is a really perfect project. I have now achieved the reproduction according to your dataset. I have a question now. I want to record my own dataset for training, but according to the readme, I feel that there is a lack of tips? (I saw your tip to check avp_teleoperate, but I don’t know how to quickly record data?) What code can I use to record the joint information and image information of the avp_teleoperate project at the same time?

ltf1001, 2025-02-12:
@silencht How should I solve the problem of collecting data code?

silencht, 2025-02-21:
Sorry for the delayed reply. There is a brief description in the README, as follows:

> If the --record parameter is used, Operator B can press s key in the opened "record image" window to start recording data, and press s again to stop. This operation can be repeated as needed for multiple recordings.

If you have any questions about the code, please feel free to reply here, and we will do our best to explain it to you.

closed 2025-02-27 by hengguo-boop