unitreerobotics/unitree_lerobot #6: The structure of the learning data has changed.

pollux-dblee, 2025-03-14:
Will unitree_IL_lerobot be upgraded? Or is there a way to use it in lerobot? 

Thanks.

hengguo-boop, 2025-03-18:
Sorry, I just saw this. My reply is a bit late.
Currently, the dataset in our Huggingface repository is the Lerobot v2.0 version. If you want to run the Lerobot policy, it is recommended to use the official Lerobot repository. 
Our repository currently only supports Lerobot datasets v1.6. 
A more complete version will be updated later.

pollux-dblee, 2025-03-18:
So can I get unitree g1 v1.6dataset?

Looking forward to your update, thank you for your reply.

hengguo-boop, 2025-03-18:
The data of the current v1.6 version is no longer maintained. 
Why is this so? You can take a look at this https://github.com/unitreerobotics/unitree_IL_lerobot/issues/5#issuecomment-2687072468  . 
It will be more convenient and user-friendly for you to train the policy using the official repository.
We will update and complete the iteration of this repository as soon as possible.

closed 2025-03-18 by pollux-dblee