unitreerobotics/unitree_lerobot #7: Problem occurred when training with provided datasets

Frozenkiddo, 2025-03-25:
I am using datasets from huggingface(https://huggingface.co/unitreerobotics), and I found out that the structure has been changed. Any chance I can get access to the older version? I know it might not continue to be maintained, but I really need it for a primary test in a very short period of time. Thanks a lot for your contribution! @hengguo-boop

hengguo-boop, 2025-03-25:
Do you need the dataset of Lerobot v1.6 on your end? The data of Lerobot v2.0 is more complete and supports more official warehouse policies.

Frozenkiddo, 2025-03-25:
Much appreciation for your reply. Truth is I can't train the DP policy under the instruction of this repo, and I noticed that the problem is caused due to inconsistency between the dataset versions. So I wonder if there is a possible way to address this while I still use this repo? @hengguo-boop 
> Do you need the dataset of Lerobot v1.6 on your end? The data of Lerobot v2.0 is more complete and supports more official warehouse policies.

hengguo-boop, 2025-03-25:
If you're in a hurry, you can use the official Lerobot repository to load our Hugging Face data for training.

closed 2025-03-27 by hengguo-boop