unitreerobotics/unitree_lerobot #8: how to train and deploy with v2.0 G1_DualArmGrasping_Dataset

eastflag, 2025-03-25:
I downloaded v2.0 unitreerobotics/G1_DualArmGrasping_Dataset.

I trained as follows.
$ python lerobot/scripts/train.py  policy=diffusion_unitree_real_g1  env=unitree_real_g1 dataset_repo_id=G1_DualArmGrasping_Dataset.

But result is "This repo doesn't have train folder."
Maybe this command is compatible with v1.6.

How to train and deploy to real with v2.0?

hengguo-boop, 2025-03-25:
Currently, the Lerobot dataset in this repository is from version 1.6 and does not support version 2.0 yet. I'll update this repository as soon as possible in the next two days to support the v2.0 version of lerobot.  If you're in a hurry, you can use the official Lerobot repository to load our Hugging Face data for training.

eastflag, 2025-03-26:
Thank you for updaing README. But I think the folder structure is a little strange.
On the readme says:
# Install LeRobot
cd lerobot && pip install -e .
# Install unitree_lerobot
cd .. && pip install -e .
I think the lerobot folder should be right under the root folder, but it's currently located in the unitree_lerobot folder under root.

hengguo-boop, 2025-03-26:
I have updated the README. The code I uploaded today and the README will be continuously updated and improved by me to solve related issues.

closed 2025-03-27 by hengguo-boop