unitreerobotics/unitree_lerobot #9: Real-World Testing Many Bugs !!!!!!

ltlhuuu, 2025-03-26:
The code for the G1 Real-World  Testing never runs, there are a lot of bugs, can you fix it?

hengguo-boop, 2025-03-26:
Can you describe what the problem is?

ltlhuuu, 2025-03-26:
1.[image]

This is the begining of the bugs.

2. https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L184-L185

There's something wrong with the size of this place.

3. https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L162-L163

I don't understand why it's 13:19 and 19:, in that case the size of this data is 6 and 9 respectively, I really don't understand, shouldn't it be 7, have you really run the code?

4. https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L153
I don't understand why it's :13, not is :14. I can not understand.

5.https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L228-L232

why you have two `observation = preprocess_observation(observation)`？

6. https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L246C5-L247

the index is right?

7. https://github.com/unitreerobotics/unitree_IL_lerobot/blob/5ae35d0f218f233dd029a7e096f4aa2e972fcac7/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L237-L239

Why is the action output from the network not sent to the cpu to dumpy?

**There are too many bugs to mention, can you  seriously fix some of them? That said, do you  actually test the code?**

hengguo-boop, 2025-03-26:
Apologies for the issues you raised. Regarding question 1, I will update the README later to clarify things. 
As for the other issues, due to the urgent update starting from yesterday, I've been facing some communication problems with my triple-hand setup. The version I submitted at noon (which only validated model loading and some basic logic) 

The version with fixes from real device testing hasn't been updated yet.. I will resolve the issues as soon as possible and provide an update. Sorry again!

This is indeed my mistake. I sincerely apologize.

closed 2025-03-28 by hengguo-boop