unitreerobotics/unitree_lerobot #11: How to train unitree-G1 in simulation

Frozenkiddo, 2025-03-29:
This repo is meant to train robot in real scene. However, I wanna train G1 in simulation. Is there already a simulator installed during the setup or should I use a specific one? If I am gonna use unitree-sdk, which simulation platform could it apply to? Much appreciation for your reply. @hengguo-boop

hengguo-boop, 2025-03-31:
At present, we only verify in real environments. I also have plans to include tests in simulations. I will improve this part as soon as possible.

Frozenkiddo, 2025-04-01:
I understand. Thank you for your reply.

> At present, we only verify in real environments. I also have plans to include tests in simulations. I will improve this part as soon as possible.

Frozenkiddo, 2025-04-01:
Got it. Thank again for your kindly reply. Looking forward to your new update! @hengguo-boop 
> The current version of unitree_mujoco does not support manipulation tasks. The repository does not include manipulation scenarios, and the G1 model lacks a camera. I will complete the G1 simulation as soon as possible—it's already in progress.

Frozenkiddo, 2025-04-08:
Sorry to bother you again. There are some following questions about this issue since I proceeded my project. If I build my own manipulation scenarios in mujoco, and added camera onto the robot in sim, is this script (_unitree_lerobot/eval_robot/eval_g1/eval_g1.py_) gonna work or it needs some modification? @hengguo-boop

hengguo-boop, 2025-04-09:
Yes, it needs to be changed, but it's quite simple. 

During the real test, the status and images were obtained from the real robot and then the actions were sent to the robot. 
You need to obtain the status and images from the simulator and then send the action execution to the simulator. 

You can refer to [lerobot's eval](https://github.com/huggingface/lerobot/blob/725b446ad6e15499c4d92acce24b72103d4ce777/lerobot/scripts/eval.py), but his is a bit more complex.

 If you want to use it quickly, you can modify and adapt the simulator data based on [eval_g1](https://github.com/unitreerobotics/unitree_IL_lerobot/blob/main/unitree_lerobot/eval_robot/eval_g1/eval_g1.py).

closed 2025-07-23 by hengguo-boop