unitreerobotics/unitree_lerobot #12: Env types missing

Frozenkiddo, 2025-03-29:
Appreciation for your brilliant team's hard work to update the repo. But I've noticed that there are certain types of env missing, like unitree-g1-env. Is it not provided now or it's just a matter of time? @hengguo-boop

hengguo-boop, 2025-03-31:
The "unitree-g1-env" type is missing. I didn't quite understand your meaning. Is this related to the [G1 simulation](https://github.com/unitreerobotics/unitree_IL_lerobot/issues/11) you mentioned?

Frozenkiddo, 2025-03-31:
Yes. Sorry for not being clear enough. What I tryna explain is that in the original lerobot repo, there are envs provided such as aloha and pusht. So I am wondering if there are similar envs to use in this repo? Thank you for your reply. @hengguo-boop 
> The "unitree-g1-env" type is missing. I didn't quite understand your meaning. Is this related to the [G1 simulation](https://github.com/unitreerobotics/unitree_IL_lerobot/issues/11) you mentioned?

hengguo-boop, 2025-03-31:
It has not been implemented in the repository yet, but I will try and implement it as soon as possible. This is also part of our plan.