unitreerobotics/unitree_lerobot #13: Robot Configuration Add

jdj1403, 2025-03-31:
Hello, I’d like to create a new issue on GitHub. I hope you can kindly assist me with this.

I’m trying to convert teleoperation data into the lerobot format. The conversion code is located at `unitree_lerobot/utils/convert_unitree_json_to_lerobot.py`. However, I noticed that the Config values for the hand robot are limited to `Z1_Dual`, `G1_Gripper`, and `G1_Dex3`. I’m wondering if it would be fine to create a new `RobotConfig` if I were to use the Inspire Hand. If additional development is required, I’d like to know when it might be scheduled for release.

Thank you!

hengguo-boop, 2025-03-31:
Our code supports custom types. You need to add your own [ROBOT_CONFIGS](https://github.com/unitreerobotics/unitree_IL_lerobot/blob/main/unitree_lerobot/utils/convert_unitree_json_to_lerobot.py#L154C1-L155C1 ) corresponding to your type.

jdj1403, 2025-03-31:
Thanks for replying.

In that case, would it be okay to directly modify `eval_robot/eval_g1/robot_control/eval_g1.py` and `eval_robot/eval_g1/robot_control/eval_g1_dataset.py` to match accordingly?

hengguo-boop, 2025-03-31:
The control for Inspire Hand has not been added. You need to implement this part yourself. You can refer to the `eval_robot/eval_g1/robot_control/eval_g1.py` and send the end-effector actions to Inspire Hand.

closed 2025-03-31 by jdj1403