unitreerobotics/unitree_lerobot #14: 实机运行一直提示Waiting to subscribe dds

cabagebird, 2025-04-03:
您好，我的机器人型号是**G1 EDU**。
问题一：根据readme，我训练好了模型，并尝试进行实机测试，但是在运行过程会重复提示 `[Dex3_1_Controller] Waiting to subscribe dds`。

问题二：根据readme中的链接[https://github.com/unitreerobotics/avp_teleoperate?tab=readme-ov-file#31-%EF%B8%8F-image-server](url)
尝试在G1内启动AVP库中的Image Server会提示：
`[ WARN:0] global ../modules/videoio/src/cap_v4l.cpp (887) open VIDEOIO(V4L2:/dev/video0): can't open camera by index
[Image Server] Camera 0 Error: Failed to initialize the camera or read frames. Exiting...`
上述问题该如何去解决？

感谢您的解答！

le-wei, 2025-04-03:
问题1：这个问题是由于手的dds没有连接上导致的。请检查网络质量以及波动情况；可以尝试调换以下手臂和手初始化的顺序可以改善情况但是无法根治。
问题2：请查看相机是否松动，以及相机的接口是否可用，并且确定插入的设备号；如'head_camera_id_numbers': [0],                                    # '/dev/video0' (opencv)    。修改head_camera_id_numbers使其对应

closed 2025-04-23 by hengguo-boop

cabagebird, 2025-05-19:
非常有用，感谢解答！