unitreerobotics/unitree_lerobot #16: ask for running eval_g1.py

IvyerS, 2025-04-16:
hi,thanks to your work, i am learning from this work, I would like to ask, does this eval_g1.py have to be implemented on the user's computer? Have you tried it on the jetson computer on the robot body? Is it feasible? Are there any hardware requirements? Do I have to be able to access the dataset on huggingface to implement it, or can I use the local dataset collected on jetson and deploy it on jetson using the policy trained on the 4090 on my host computer?

hengguo-boop, 2025-04-23:
I'm very sorry for seeing the message now. 
We haven't tried deploying the model on Jeston (you can try deploying it after acceleration with TensorRT). Currently, we are all deploying it on 4090.

IvyerS, 2025-04-28:
ths, i will test it on Jeston

closed 2025-05-12 by IvyerS