unitreerobotics/unitree_lerobot #18: 如果要适配因时灵巧手，应该怎么做？

zjukop, 2025-05-20:


hengguo-boop, 2025-05-21:
Sorry for the late reply. 
Currently, we don't have recorded data for the dexterous hand. 
So, the first step for you should be to [record your own data corresponding to the inspire hand](https://github.com/unitreerobotics/avp_teleoperate/blob/main/teleop/robot_control/robot_hand_inspire.py). 
After the model is trained, 
for deployment, you just need to [replace our three-finger hand control with the inspire hand control](https://github.com/unitreerobotics/unitree_IL_lerobot/blob/02063f430c0fd327648e68d9c46b0478f58bcf09/unitree_lerobot/eval_robot/eval_g1/eval_g1.py#L130).

closed 2025-06-11 by hengguo-boop

PEAGREEABLB, 2025-07-22:
有计划更新unitree_IL_lerobot的项目以支持inspire灵巧手吗

PEAGREEABLB, 2025-07-22:
> 有计划更新unitree_IL_lerobot的项目以支持inspire灵巧手吗

我在将xr_teleoperate中的inspire控制部分移植到本项目中，但是问题有点多