unitreerobotics/unitree_lerobot #20: Real World Testing is Not Working

HarunEnsar, 2025-06-13:
Hi, 

### Description
I trained the policy using the G1_BlockStacking_Dataset without modifying the unitree_lerobot repo. In real-world testing, the robot does not attempt to pick up the cubes. It seems stuck in a loop of minimal movements.

### What I Did
- Followed installation and setup steps from README.
- Verified camera input is working (both head and wrist).
- Used the following command to run evaluation:

```bash
python unitree_lerobot/eval_robot/eval_g1/eval_g1.py  \
    --policy.path=/home/ltcai2/unitree_IL_lerobot/outputs/train/2025-06-01/19-10-08_act/checkpoints/040000/pretrained_model \
    --repo_id=unitreerobotics/G1_BlockStacking_Dataset

hengguo-boop, 2025-06-16:
I suggest re-recording some of your own data and then training and testing it, 
because the current imitation learning algorithms are not very universal for data in different environments.

hengguo-boop, 2025-06-16:
1. How the data is distributed, and data at different positions can be added.
2. It is recommended to train for about 50 epochs.
3. When validating the model, it is necessary to check whether the input of the validation data is consistent with that of the training data.
4. Also, in my code, there is a part where the dataset is loaded to compare the action in the dataset with the action output by the model. You can check the difference in the predicted actions of the model. If the predicted actions of the model do not differ much from the actual actions in the dataset, it is necessary to check the hardware of the robotic arm.

closed 2025-07-10 by hengguo-boop

PEAGREEABLB, 2025-07-24:
HI，Did you solve this problem？I also had the same problem, would like to refer to your solution below, thank you！

soneo1127, 2025-08-20:
Looking at the issues, it seems no one has managed to run the training results on the actual hardware. Has anyone other than the official team been able to do it?

Astro-Q, 2025-11-11:
> 从目前的情况来看，似乎没有人成功地在实际硬件上运行过训练结果。除了官方团队之外，还有其他人做到过吗？

请问您有运行起来吗？我还在努力尝试