unitreerobotics/unitree_lerobot #30: Wireless Communication with G1 EDU Robot and Running eval on Development Unit

HarunEnsar, 2025-08-19:
Hi,

I have a question regarding testing trained models on the G1 EDU robot.

Currently, the examples and documentation show communication via Ethernet.

1. Is it possible to communicate with the G1 EDU robot wirelessly (via Wi-Fi) using the SDK?
For our setup, a wireless solution would be preferred — ideally without requiring a wired switch.

2. Alternatively, is it possible to install the project and run the evaluation scripts (e.g., eval_g1.py) directly on the development computing unit (PC2) inside the robot?

This would avoid the need to run evaluation from another external PC.

hengguo-boop, 2025-08-20:
1.The Wi-Fi mode can be used, but the data transmission is definitely not as stable as the wired model. 
2.PC2 is an NVIDIA NX. If your model is relatively small and has been accelerated, it can be fully verified on PC2. If your model is large, it is recommended to handle it through client-server processing, inferring the model on the host and then sending it to PC2.

HarunEnsar, 2025-08-24:
1. Is there any guide or documentation available for using the SDK over Wi-Fi? Are there detailed steps explaining how to set up a Wi-Fi connection? 

2: This repo requires Python 3.10, but for G1 PC2 (JetPack 5.1 (Orin NX, CUDA 11.4)) there is no officially or community-provided PyTorch wheel with CUDA support for Python 3.10 — after long research I couldn’t find any.

silencht, 2025-08-25:
Hello,
> Is there any guide or documentation available for using the SDK over Wi-Fi? Are there detailed steps explaining how to set up a Wi-Fi connection?

There are many documents about enabling Wi-Fi on the Jetson NX development board, such as [this one](https://support.unitree.com/home/en/G1_developer/FAQ)
 and [this one](https://global-serviceconsole.unitree.com/#/help/031303). However, simply turning on Wi-Fi does not enable robot control using the SDK. Here is the network topology diagram of the robot:

[image]

As you can see: the wireless NIC of PC2 is not on the same LAN as the wired network of PC1. Therefore, without additional measures, it cannot communicate with PC1 via DDS. You can also refer to this document [CycloneDDS and UnitreeSDK](https://github.com/unitreerobotics/xr_teleoperate/wiki/CycloneDDS-and-UnitreeSDK-(en))
 for a deeper understanding of how DDS communication works.

I have previously tested bridging PC2’s wireless NIC with the wired NIC at the data link layer, but the process showed that DDS communication was not stable.

Perhaps you could try solving this problem through network-layer proxy routing/forwarding or application-layer DDS topic relaying. After studying the above documentation, you might also want to consult ChatGPT — it could provide suggestions on these approaches.

> This repo requires Python 3.10, but for G1 PC2 (JetPack 5.1 (Orin NX, CUDA 11.4)) there is no officially or community-provided PyTorch wheel with CUDA support for Python 3.10 — after long research I couldn’t find any.

You can also contact technical support to obtain instructions on upgrading to the JetPack 6.2 package.