unitreerobotics/unitree_lerobot #32: 真机测试时候，为何手臂和灵巧手的动作陷入微弱的动作循环中？请问要如何排查呢？

dailouer-guo, 2025-08-22:
[image]

[image]

我们经过遥操采集数据后（https://github.com/unitreerobotics/xr_teleoperate），进行训练100K steps，但是真机测试的结果是：手臂和灵巧手都是一些微弱的动作，循环往复，请问这个问题要如何排查呢？

dailouer-guo, 2025-08-25:
> 1.尝试replay数据 确定摇操采集的数据没有问题 。 2.检查模型输入输出是否一致，输入的颜色通道。 3.模型训练可以多训练，40~50 epoch。 4.数据太少也可能学不会（对于act dp需要比较多的数据）

感谢回复，对应还有以下疑问：
1、如何repaly？是执行可视化这个命令查看吗？python src/lerobot/scripts/visualize_dataset.py \
    --repo-id unitreerobotics/G1_ToastedBread_Dataset \
    --episode-index 0
2、如何设置epoch？我看lerobot这里没有能设置epoch的参数，但是提高step的时候，epoch会增加，是要直接增加step吗？step默认100k。
3、数据太少是指遥操作采集的数据太少吗？对于一个抓取物品到指定位置的任务，大概需要采集多少ep数据来训练？

hengguo-boop, 2025-08-25:
1.replay可以直接将数据集的action发送到机器人，查看replay的数据质量
2.训练输入应该有ep的参数(终端中训练输出) 可以对应增加相应的step
3.是的，假如训练到基本过拟合效果不好就可以增加数据了

onestardao, 2025-08-25:
抱歉打扰一下，我看了一下这里的情况。
这个现象多半还是出在 数据分布和 replay 缓存一致性，而不仅仅是参数设置。
如果采样分布偏差，会导致模型在真机上进入弱循环，哪怕 epoch/step 数再调也难以稳定。
建议除了参数尝试，也特别检查一下 replay 质量与数据多样性。
仅供参考，希望对你们排查有帮助。

zmingen, 2025-08-27:
想要训练一个pick and place的动作，大概需要什么多少数据量？

hengguo-boop, 2025-08-28:
简单任务 100~200条数据 可以

dailouer-guo, 2025-08-29:
> 抱歉打扰一下，我看了一下这里的情况。 这个现象多半还是出在 数据分布和 replay 缓存一致性，而不仅仅是参数设置。 如果采样分布偏差，会导致模型在真机上进入弱循环，哪怕 epoch/step 数再调也难以稳定。 建议除了参数尝试，也特别检查一下 replay 质量与数据多样性。 仅供参考，希望对你们排查有帮助。

感谢建议，我这边确实是测试了数据的replay，个别数据确实是按照原来轨迹抓取物体的。
但是采集了389条动作数据，训练43个ep，效果是：模型推理开启后，手部移动到指定位置，灵巧手有动作但是始终无法抓起物体，然后只是将瓶子推到指定位置，而非抓取。

[image]