unitreerobotics/unitree_lerobot #36: 训练时候偶然会有报错，发现转换的视频帧有问题，请问是BUG吗？

dailouer-guo, 2025-09-02:
步骤如下：
 1、python unitree_lerobot/utils/convert_unitree_json_to_lerobot.py \
    --raw-dir $HOME/datasets \
    --repo-id your_name/repo_task_name \
    --robot_type Unitree_G1_Inspire

2、python src/lerobot/scripts/train.py \
    --dataset.repo_id=unitreerobotics/G1_ToastedBread_Dataset \
    --policy.push_to_hub=false \
    --policy.type=act 

发现训练过程中会报错：
INFO 2025-09-01 11:57:19 ts/train.py:232 step:22K smpl:179K ep:311 epch:0.61 loss:0.145 grdn:7.339 lr:1.0e-05 updt_s:0.151 data_s:0.000
INFO 2025-09-01 11:57:50 ts/train.py:232 step:23K smpl:181K ep:314 epch:0.61 loss:0.142 grdn:7.042 lr:1.0e-05 updt_s:0.155 data_s:0.000
Traceback (most recent call last):
  File "/home/hp/G1/VR/lerobot/src/lerobot/scripts/train.py", line 295, in <module>
    main()
  File "/home/hp/G1/VR/lerobot/src/lerobot/scripts/train.py", line 291, in main
    train()
  File "/home/hp/G1/VR/lerobot/src/lerobot/configs/parser.py", line 225, in wrapper_inner
    response = fn(cfg, *args, **kwargs)
  File "/home/hp/G1/VR/lerobot/src/lerobot/scripts/train.py", line 205, in train
    batch = next(dl_iter)
  File "/home/hp/G1/VR/lerobot/src/lerobot/datasets/utils.py", line 630, in cycle
    yield next(iterator)
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 733, in __next__
    data = self._next_data()
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1515, in _next_data
    return self._process_data(data, worker_id)
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/dataloader.py", line 1550, in _process_data
    data.reraise()
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/_utils.py", line 750, in reraise
    raise exception
RuntimeError: Caught RuntimeError in DataLoader worker process 2.
Original Traceback (most recent call last):
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/worker.py", line 349, in _worker_loop
    data = fetcher.fetch(index)  # type: ignore[possibly-undefined]
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/utils/data/_utils/fetch.py", line 52, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "/home/hp/G1/VR/lerobot/src/lerobot/datasets/lerobot_dataset.py", line 720, in __getitem__
    video_frames = self._query_videos(query_timestamps, ep_idx)
  File "/home/hp/G1/VR/lerobot/src/lerobot/datasets/lerobot_dataset.py", line 692, in _query_videos
    frames = decode_video_frames(video_path, query_ts, self.tolerance_s, self.video_backend)
  File "/home/hp/G1/VR/lerobot/src/lerobot/datasets/video_utils.py", line 66, in decode_video_frames
    return decode_video_frames_torchcodec(video_path, timestamps, tolerance_s)
  File "/home/hp/G1/VR/lerobot/src/lerobot/datasets/video_utils.py", line 206, in decode_video_frames_torchcodec
    frames_batch = decoder.get_frames_at(indices=frame_indices)
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torchcodec/decoders/_video_decoder.py", line 227, in get_frames_at
    data, pts_seconds, duration_seconds = core.get_frames_at_indices(
  File "/home/hp/miniconda3/envs/unitree_lerobot/lib/python3.10/site-packages/torch/_ops.py", line 756, in __call__
    return self._op(*args, **kwargs)
RuntimeError: Could not push packet to decoder: Invalid data found when processing input 

找到对应的视频，发现有些异常：

[image]

但是我采集了388条数据后，后续在此基础上采集到了511条，都是同样的episodes，使用388的数据集转换没这个问题，但是使用511转换就会出现这个异常视频，请问是BUG吗？需要如何解决？

hengguo-boop, 2025-09-02:
感觉应该是录数据图片转到视频保存转换的问题吧 你可以确认下录数据时图片质量有没有问题  
重新从图片转呢 有试过吗？这个问题第一次出现还是之前也出现过。

dailouer-guo, 2025-09-02:
> 感觉应该是录数据图片转到视频保存转换的问题吧 你可以确认下录数据时图片质量有没有问题 重新从图片转呢 有试过吗？这个问题第一次出现还是之前也出现过。

这个之前也有出现过，我们采集了一次388ep的数据集，训练时候出现过，但是我们把异常数据替换后，转换后也能继续训练，但是将数据集从388增加到511（包含了之前388条数据），转换以后，训练的时候报错：RuntimeError: Could not push packet to decoder: Invalid data found when processing input。打开视频发现，之前由511个数据集转换后的第125和第243个视频数据就会有上面的异常视频，但是在388个数据集转换后的，同样是第125和第243个就没有这个异常。

hengguo-boop, 2025-09-02:
确认过了保存的图片的质量没有问题是吧？
重新转一次看看是不是还会有同样的问题

dailouer-guo, 2025-09-02:
> 确认过了保存的图片的质量没有问题是吧？ 重新转一次看看是不是还会有同样的问题

感觉是偶发的，我单独将这两个数据抽出来转换，也没有这个问题。